Mobsya/aseba #723: studio play and stop pause step bouton behavior

mbonani, 2019-11-21:
[image]

It is strange that pause and step button disappear when robot is not in play mode, it make the play and stop button also change place.
It suggest that all button are always shown and put in gray level when no active. 
Also if a correct code is there we should let play button be active even if robot is not stop.

cor3ntin, 2019-11-25:
> Also if a correct code is there we should let play button be active even if robot is not stop.
What does clicking that button is supposed to do then?

mbonani, 2019-11-25:
load and run the valide code, you just don't need to stop it before, like in VPL you can run even if robot is not stop before.

cor3ntin, 2019-11-26:
That is very weird UI wise because it looks like pressing play does nothing

mbonani, 2019-11-26:
if the code is valide, it should resend it and make the robot start again so play button has always an action, why should the user stop the robot to send the new code?

mbonani, 2019-11-27:
fix in PR #724

closed 2019-11-27 by mbonani