Mobsya/aseba #729: laucher let connection to Wireless Thymio not fully describe for aseba studio

mbonani, 2019-11-28:
If its possible to detect that robot are still not fully describe, do not let studio connect to them.
Connection to already use robot should be permit.

riedo, 2020-03-11:
One suggestion: the button that says "lancer Aseba studio" could state different things depending on the case. It could say something like "connect and program" if the robot is available or "watch" if it is already taken by another user or before actual bind. 
It makes it explicit for the users and if they want to program they can wait until the button is the right one (and in VPL etc, the button could be greyed out until connection is actually possible)

mbonani, 2020-03-13:
suggestion was implemented and bug is fix in 18baaf06531c0e59b0ededa10ba12d8df8047e2b

closed 2020-03-13 by mbonani