Mobsya/aseba #731: VPL sizing windows create sometimes bad proportion for the buttons

mbonani, 2019-12-02:
[image]

mbonani, 2019-12-02:
it happen at least on  Macosx 10.14,

mbonani, 2019-12-04:
on a 10.14.3 it is working
[image]
[image]

mbonani, 2019-12-04:
it works on a 10.14.3 
[image]
[image]

mbonani, 2019-12-04:
it do not work on another 10.14.x

> La résolution d’écran n’a pas d’importance. La taille verticale de la fenêtre a un minimum. Tu va à ce minimum, puis tu élargis la fenêtre. Sur un écran 
> J’arrive à voir le problème avec un écran de 1680 de large, 
[image]

mais avec un écran de 3008 de large les icônes disparaissent

[image]

mbonani, 2019-12-04:
I understand better the calclulation of the is wrong when you have a small high and a big with
[image]

podak, 2020-01-20:
This issue seems to be generated by a combination of how Qt manages the resizing of elements inside QGridLayout containers and how this layout can be combined with others inside the application. In particular I tried the following things:

- Set a minimum height to both the rows of the GridLayout and the PushButton in the cells. This successfully stopped the container elements to reduce their height, but the inner icon keeps resizing until it disappear
- Set a fixed dimension to the inner icon with the setIconSize function. This apparently had no effect and the icons keep resizing.
- Isolate the problem skipping the rendering of the action/event area ( on the bottom ) and the aseba code editor ( on the right ). This somehow deactivate the defect and the icons do not disappear anymore

mbonani, 2020-04-02:
actual version 
[image]

closed 2020-04-22 by mbonani