Mobsya/aseba #737: webgl problem on some win 10 computer

mbonani, 2020-01-07:
Scratch 3 do not launch inside our webtool using chromium (on the BETA 2)  showing a webgl problem:
[image]

users say that webgl is working on is normal Chrome and computer has normally updated drivers also for the graphic card.
[image]

mbonani, 2020-01-08:
Another user with a 32 bit version:

> J’ai installé Thymio Suite sur un Dell Optiplex360 en dual boot Linux Mint 19 xfce et windows 7 afin que mes élèves puissent programmer Thymio avec Scratch.
> Sous Mint cela fonctionne bien en revanche j’ai le message « Votre navigateur ne supporte pas WebGL » sous W7. Le problème vient donc de ma carte graphique cependant la version Scratch Desktop 3.6 fonctionne sans problème. Serait-il possible d’intégrer cette version dans Thymio Suite ?

mbonani, 2020-01-27:
we have more and more feedback around this problem, probably updating or configurating the chromium should be sufficient

KunjanChauhan, 2020-02-17:
Status agreed at 2020 02 17 meeting: 

- [ ] 1. Valentina/Davide to enable Chrome address field so Fanny can test the inbuilt Chrome browser to see if this is the source of the problem
- [ ] 2. in parallel, Valentina to think about other sources of this bug

riedo, 2020-02-18:
could be this:
https://bugs.chromium.org/p/chromium/issues/detail?id=1036367

riedo, 2020-03-03:
FYI, some users told me this:

**User 1**
On his computer, he has the webgl error in our embarked browser, but it works fine in his firefox
intel core 2 Duo CPU E7300
Windows 10 64 bits
Firefox Quantum 69.0.1 (64 bits) with settings:
webgl.enable-webgl2 true
webgl.force-enable true

**User 2**
Scratch works in Edge 80.0.361.54, but he gets the webgl error in our embarked browser

podak, 2020-03-10:
With the first pull request we enabled a text field on the top of the browser with the url of the page. This solution did not work on Windows ( caused the crash of the included browser ), also the look of the text bar did not meet the expectations.
After discussing with Michael, we will propose a fix following a different approach: a checkbox will be added to the launcher, the user will be able to select if use the included browser or the system default one

podak, 2020-03-12:
After further discussion we decided to move the checkbox on the collapsable right menu of the main page in thymio suite

mbonani, 2020-03-13:
the checkbox is implemented in 9ab73f2

closed 2020-03-13 by mbonani