Mobsya/aseba #743: changing the languages in blockly make you connected to all free robots robot

mbonani, 2020-01-31:
when you change the language in Blockly it then connect to all other robot and send code probably to another one.

ValentinaTibs, 2020-04-02:
How to know/check which is the robot to whom Blockly is connected to?

mbonani, 2020-04-03:
there is no way directly but when you change languages you see it in fact that the address bar is not keeping the parameters: 
file:///C:/Program%20Files/Thymio/thymio_blockly/thymio_blockly.en.html#device=%7B005207f4-5a8c-4374-bb66-08924ff5a057%7D&ws=ws://192.168.1.5:8597
then when I put in French adress is: 
file:///C:/Program%20Files/Thymio/thymio_blockly/thymio_blockly.fr.html
So the bug is not in Thymio suite but in Blockly.
also in the Suite you see all robots are used because of the default adress. 
@podak  I think this is more something you know how to fix?

ValentinaTibs, 2020-04-15:
this is the pure vanilla javascript code lines that would solve the problem  if putted in thymio_blockly.js
I didn't get the building pipeline for the nodejs part of the custom blockly fork 
so here they go  - we can also add them into the specific languages raw .html files  that are managed independently from the blockly interface as the info button or similar
such that
[https://github.com/Mobsya/thymio-blockly-standalone/blob/master/thymio_blockly/thymio_blockly.it.html
](url)
```
<script type="text/javascript">
        
document.addEventListener('DOMContentLoaded', function() {

const add_info = window.location.href.split(/(.html)/)[2]
          var container = document.getElementsByClassName("lang");
          for (var i = 0; i < container.length; i++) {
            var links = container[i].getElementsByTagName('a');
            for (var j = 0; j < links.length; j++) {  
               links[j].href += add_info;
             }
          }
        }, false);

      </script>
```

closed 2020-04-16 by ValentinaTibs