Mobsya/aseba #746: TDM or Thymio suite seems not work when a proxy is activated

mbonani, 2020-02-05:
previously with aseba 1.6 if you add "not use proxy for local connection" was solving the proble that is no more working now.

bergerpascal, 2020-05-27:
We are having the same problem. Is there any news on that issue?

mbonani, 2020-05-27:
Hello,
No news, we were not working on it, but could you give me more context? Sometimes it is also a firewall that blocks.
Regards

bergerpascal, 2020-05-27:
Hi Michael,
thank you for answering me.

I am just the guy that dose the packaging for SCCM.
But I got informed by the support engineer, that he had the problem that he could not connect the robot. He removed the proxy form the IE configuration and then it worked. For use this is a big problem because the schools where they want to use the robots relay on using that proxy.
Since the were using 2.0.0 RC 5 I suggested to them to test 2.0.3.
Do you have an estimate when you will investigate that issue and when it could be fixed?

Kind regards

Pascal

mbonani, 2020-05-28:
Hi put here feedback from the user: 

> Lorsqu’une adresse proxy est déclarée dans les paramètres système (ce qui est le cas dans la majorité des établissement scolaire je pense), les robots ne sont pas détéctés (ni en USB, ni avec le Dongle).
> Avec Aseba, il suffisait de cocher « ne pas utiliser de proxy pour les adresse locales » pour régler le problème de connexion mais cela ne suffit plus avec Thymio suite.

> « Ne pas utiliser le serveur proxy pour les adressse (intanet) locales » fonctionne avec aseba 1.6 mais pas avec Thymio suite. Le fait de désactiver le proxy fait que Thymio suite fonctionne correctement?

> Je vous confirme que nous utilisons un proxy manuel (c’est le cas pour toute l’académie de Strasbourg).
> Proxy déasctivé, Thymiosuite fonction parfaitement.
> 
> Nous utilisons donc pour l’instant Aseba et Scratchx en cochant « ne pas utiliser de proxy pour les adresse locales ».

mbonani, 2020-05-28:
to be evaluted

mbonani, 2020-06-08:
after looking for the Policy, which causes the problem, we found that “ProxySettingsPerUser”(you were right, it is a problem with the Proxy-Server) is the reason, it doesn’t work.

To solve the problem, we need the URL which the software connects to. If we have the URL we can tell the Proxy, that he can let it through.

mbonani, 2020-06-08:
back from lunch connection with scratch from computer or ipad do not work, it work with studio or VPL, is there something wrong with this connection link?
file:///C:/Program%20Files/Thymio/scratch/index.html?device=%7B43b62dfe-7796-41d4-a2d1-980d5aa2417b%7D&ws=

mbonani, 2020-06-08:
connection link should have IP adress like this 
file:///C:/Program%20Files/Thymio/scratch/index.html?device=%7Beead4e83-4b0d-4dd4-8fb7-56d9dbaf4332%7D&ws=ws://192.168.1.4:8597

mbonani, 2020-06-08:
ok If I correct Ip by hand on address bar adding IP from mac it works! file:///C:/Program%20Files/Thymio/scratch/index.html?device=%7B43b62dfe-7796-41d4-a2d1-980d5aa2417b%7D&ws=ws://192.168.1.3:8597

mbonani, 2020-09-03:
branch  'proxy-local' is a working solution with proxy. Thymio Suite connect directly to the local host without using Bonjour service
you can test build here: https://dev.azure.com/mobsya/Thymio%20Software%20Suite/_build/results?buildId=1682&view=artifacts&type=publishedArtifacts

mbonani, 2020-09-08:
special version released

closed 2020-09-08 by mbonani