Mobsya/aseba #760: studio open "recent" file do not work on osx

mbonani, 2020-03-12:
in studio open file normally choosing from finder works, but open files from the "recent" list files do not open the files correctly (code do not appear)  test on osx Sierra 10.12

ValentinaTibs, 2020-04-01:
@mbonani I've started watching this and from my early observation to me it looks this happens just when the file is not present in the filesystem anymore. 
Can you confirm that the file was still present in the recent file list but not in the filesystem?

mbonani, 2020-04-01:
I cannot test on the computer I have seen the bug. I make the first test on my actual windows and it seems the behaviour is the same as on the mac when the file is missing from the system.  So it is most probable that the bug is only when the file is missing.

ValentinaTibs, 2020-04-02:
Confirmed to happen in the following scenario:

1. create a new file 
2. close the editor
3. open the editor
4. select the file from the recent for opening

ValentinaTibs, 2020-04-02:
Not confirmed on Catalina 
[image]

mbonani, 2020-04-03:
confirm on 10.14 .4
Is it because the computer are in another languages then EN? so path are translated and it do not correspond on the disk?

mbonani, 2020-04-09:
fix in latest commit

closed 2020-04-09 by mbonani