Mobsya/aseba #775: Connecting pyThymio to asebaplayground via asebamedulla

HamidiMassinissa, 2020-04-16:
I want to connect [`pyThymio`](https://github.com/pierreboudes/pyThymio) to `asebaplayground` via `asebamedulla`.

I launch assebalayground with a predefined playground as follows:
`asebaplayground thymio-discovery-yeti/world.playground`

I, then, run the switch to create a link via D-Bus as follows:
`asebamedulla -p 33334 -v "tcp:localhost;33333"`

Everything is going fine till here: asebamedulla is connected to asebaplayground as depicted in the following figure.
<p align="center">
[image]
</p>

And we have d-bus service that is registered as you can see in this figure:

<p align="center">
[image]
</p>

Therefore, I should supposedly get something happening when I run [this](https://github.com/pierreboudes/pyThymio/blob/master/motors.py) snippet, however nothing happens.

Any ideas? is the procedure, in iteself, sound? @pierreboudes, any insights? I am seeking help to aleviate other, potentially long, hours of debugging.

mbonani, 2020-04-16:
you can probably launch asebameddula directly in the playground scene changing last command inside the .playground :  
	`<process command=":asebamassloader --once thymio-default-behaviours.aesl tcp:localhost;33333" />`
into 
	`<process command="asebamedulla -p 33334 -v tcp:localhost;33333" />`

HamidiMassinissa, 2020-04-17:
Many thanks for your reply @mbonani 
Indeed, your solution is straightforward and will greatly simplify the installation procedure for students.
The problem though is related to the second part of the story, which is making the robot move and send back information in the form of events to a python program. It seems that there is no communication between [this](https://github.com/pierreboudes/pyThymio/blob/master/motors.py) snippet (provided in the original post) and the playground.
Precisely, given the following code, I am waiting for events of the type "fwd.motor" but it seems that they are not captured.

```python
    def dispatch(evtid, evt_name, evt_args):
        # https://www.thymio.org/en:thymioapi motor freq is 100Hz
        if evt_name == "fwd.motor": # every 0.01 sec
            state["left"] += [evt_args[0]]
            state["right"] += [evt_args[1]]
            if state["time"] in state["prog"]:
                (l,r) = state["prog"][(state["time"])]
                Thym.set('motor.left.target', [l])
                Thym.set('motor.right.target', [r])
            state["time"] += 1
            if state["time"] == state["end"]:
                Thym.stop()

        else: # Wat?
            print evt_name
```
I greatly appreciate your help.

mbonani, 2020-04-17:
I never use pyThymio and it is not one of our development. What I just understand is that the pythymio send a specific code to Thymio loading an aesl depending on used sensors or events [https://github.com/pierreboudes/pyThymio/blob/10017bd1da28463fb5689d5f1789c88c500f4c83/pythymio.py#L85]() then Thymio should send events. The generated aesl code should have somewhere something like this 
`onevent motor`
`emit fwd.motor [motor.left.speed,motor.right.speed]`

I also do not know if all features of real Thymio used by pythymio are in the simulator. 

I cannot help more.

HamidiMassinissa, 2020-04-21:
Thanks for your response.
One question about `asebastudio`: is it listening to `asebaplayground` events in a similar way, i.e. via d-bus ? I actually connect studio to playground and I get, for example values of `prox.horizontal` when the robot is approaching the wall, from the playground.

mbonani, 2020-04-22:
There is two kinds of events in aseba. First are system events that are sent by client (studio, whatever connected to dbus), to get information or send code to the nodes (robots). In 1.6.1 Studio is asking the value for the proximity and all other value with that but at a slow rate. 
The second type are user/global events that each node can send depending on the code that their are executing, like in the previous code with the `emit` command. So as I understand the myPython it seems that depending on what is used it generate a code .aesl that is then sent to the robot that then send back events with useful value using second type of messages

mbonani, 2021-04-23:
can we close this issue?

closed 2021-12-13 by mbonani