Mobsya/aseba #789: Interface of Aseba Studio is blinking

ldbk, 2020-05-01:
Hello.
I'm using Aseba Studio on Ubuntu 18.04 (the 2.0.2 version), and the interface is constantly blinking in different windows (in the windows Variables or Natives functions: if an item is clicked, another is underlined). Writing code is quite impossible: the character typed is appearing with a lag of a few second, and blink.  The same behaviour appears with the VPL interface and in this case, I cannot use it at all to program.

mbonani, 2020-05-07:
Hello,
Strange behaviour, do you have such problems with another programs not related to Thymio suite? For me it seems a graphic card problem. 
How is reacting scratch 3? 
Is there in studio a message that the robot is not connected?
Kind regards

ldbk, 2020-05-07:
Unfortunately, since the update to the version 2.0.2, the thymio suite program is empty, and I cannot repeat this behaviour. Scratch 3 was working fine, and a thymio was connected (both real and virtual: I've tried the two options). Regarding the hardware, I'm using an Intel Xeon(R) E-2176M CPU @ 2.70GHz × 12 with Intel HD Graphics (Coffeelake 3x8 GT2).
I hope this help.
I will try to investigate the new problem first before reporting more precisely the blinking behaviour.

Regards

mbonani, 2020-05-07:
Is there a problem with the flathub/flatpak installation? Perhaps try to uninstall first using :
`flatpak uninstall org.mobsya.ThymioSuite`