Mobsya/aseba #807: Scratch and Blockly couldn't be used on Mac since RC6

ramuntcho, 2020-05-20:
Hi, 
We wanted to install ThymioSuite in our Mac labs in Geneva (Switzerland). Everything was fine with RC6, Scratch could be used (in a window Scratch 3.0 GUI) and also Blocky (in a window named Thymio Blockly).

But with ThymioSuite 2.0, 2.0.2 ,2.03, a click on the button for Scratch launches Safari and try to open file:///private/var/folders/s4/hksdz_8x1wz0h3m12jxcjkqh0000gn/T/org.mobsya.ThymioLauncher/aMyfAm.html. And nothing ... I have had a look in the folder org.mobsya.ThymioLauncher and there isn't aMyfAm.html file
Same thing with Blockly, Safari launches and try to open file:///private/var/folders/s4/hksdz_8x1wz0h3m12jxcjkqh0000gn/T/org.mobsya.ThymioLauncher/vEQyom.html but vEQyom.html doesn't exist.

So why this diffrence between RC6 and the 2.0.x version, why Safari is launched ?

We continue to use version RC6 on our Mac labs for the time being.

mbonani, 2020-05-20:
Hello,
There was an issue with saving file with some OSX  machine. We introduce the possibility to let user choose either the system default browser (we encourage Chrome or Firefox) or use the include Chromium (Include in QT) inside the app. 
For that you can uncheck the option in the tools:
[image]

mbonani, 2020-05-20:
what is you version of OSX?

ramuntcho, 2020-05-20:
Hi,

macOS 10.13, 10.14 and I think my colleague test it in 10.15.
________________________________
Christian Oïhénart
SEM-Logistique
Etat de Genève - Service Ecoles-Médias
Rue des Gazomètres 3 - CP<x-apple-data-detectors://0/0> 241 - 1211 Genève 8
Tél. 022 388 63 76<tel:022%20388%2063%2076>
christian.oihenart@edu.ge.ch<mailto:christian.oihenart@edu.ge.ch>

ramuntcho, 2020-05-23:
Thanks for the answer, i didn't see this option. It could be great that if you uncheck the option, the choice is saved when you close Thymio.

closed 2020-10-02 by mbonani