Mobsya/aseba #823: Scratch and Blockly are not working on Rasberry Pi 4

Ozon2, 2020-06-22:
Hello,
I followed the compilation instructions on a Rasberry Pi 4 for Thymio Suite 2.0, VPL and Aseba are working fine but i get an error when I try to launch Scratch or Blockly saying that the site can't be reached. Blockly is working fine with https://github.com/Mobsya/thymio-blockly-standalone but not on Thymio Suite.
[image]

mbonani, 2020-06-23:
Hello,
The link we see inside is for me not correct, on windows it is something like this: `file:///C:/Program%20Files/Thymio/scratch/index.html?device=%7B177cf107-e5a0-46d5-aa69-ecfd18fb9cac%7D&ws=ws://192.168.1.3:8597`
Do you have test our flatpak? https://flathub.org/apps/details/org.mobsya.ThymioSuite
On Linux normally we use also the webengine of QT, so why it want to open in your local browser?

Ozon2, 2020-06-23:
I can't find this index.html file anywhere, here is the output i get when the error happens :
```
QUrl("/index.html?device=%7Bb0cba2cc-d3fa-42f7-85a1-a18e3af08cb3%7D&ws=ws://192.168.0.45:8597")
qrc:/qml/ThymioSelectionDeviceDelegate.qml:222:26: Unable to assign null to int
qrc:/qml/ThymioSelectionDeviceDelegate.qml:280:17: Unable to assign QQuickColumn to QQuickAnchorLine
Searching for  "asebaplayground"
"asebaplayground" found:  "/home/pi/aseba/build/bin/asebaplayground"
qrc:/qml/ThymioSelectionDeviceDelegate.qml:28: TypeError: Type error
Resource :/index.html not found or is empty
```

The flatpak doesn't work either, i get this error :
```
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
Attribute Qt::AA_ShareOpenGLContexts must be set before QCoreApplication is created.
Searching for  "thymio-device-manager"
"thymio-device-manager" found:  "/app/bin/thymio-device-manager"
thymio-device-manager starting
Attribute Qt::AA_EnableHighDpiScaling must be set before QCoreApplication is created.
"fr_FR" "launcher_" ":/translations"
"fr_FR" "qtdeclarative_" "/usr/translations"
Didn't load translation "qtdeclarative_"
"fr_FR" "qtbase_" "/usr/translations"
Didn't load translation "qtbase_"
"fr_FR" "qtwebengine_" "/usr/translations"
Didn't load translation "qtwebengine_"
qrc:/qml/SettingsMenu.qml:132:21: QML QQuickItem: Possible anchor loop detected on vertical anchor.
qrc:/qml/SettingsMenu.qml:132:21: QML QQuickItem: Possible anchor loop detected on vertical anchor.
qrc:/qml/SettingsMenu.qml:132:21: QML QQuickItem: Possible anchor loop detected on vertical anchor.
qrc:/qml/SettingsMenu.qml:132:21: QML QQuickItem: Possible anchor loop detected on vertical anchor.
Unable to find target for this triple (no targets are registered)
```
I have no idea why it wants to open in my local browser.

THS-on, 2020-07-15:
I updated the AUR package today and might found the issue.

The build instructions are missing some instructions.

0. Clone aseba with all submodules
1. Download `https://github.com/Mobsya/scratch-gui/releases/` latest release and extract it into the main directory with the name `scratch`
2. Download `https://github.com/Mobsya/thymio-blockly-standalone/releases` latest release and extract it into the main directory with the name `thymio_blockly`
3. Download `https://github.com/Mobsya/ci-data/releases/download/data/vpl3-thymio-suite.tar.gz latest release and extract it into the main directory with the name `tvpl3-thymio-suite`.
4.  `mkdir build && cd build`
5. `cmake  -DMOBSYA_WEBAPPS_ROOT_DIR=share/ -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF ..`
6. `make && make install`

This worked for me on arch.

Is there a better way to get Scratch, Blockly and VPL3?  
Where is the repository for VPL3 if I want to build it from source?

mbonani, 2020-07-15:
Thank for investigating, we will update.
VPL3 is for now in a private repository, but it should be public soon

Ozon2, 2020-07-15:
Thank you @THS-on, it solved my issue.

closed 2020-07-15 by Ozon2