Mobsya/aseba #849: Click on "Play Melody" is off

nbornand, 2020-08-24:
Using VPL3
Clicking on a music note often modifies the note right of the click target.

mbonani, 2020-08-24:
Cannot reproduce
Can you give us on wich system/OS you are working?
Kind regards

nbornand, 2020-08-26:
Using Windows 10 Pro, version 1903 build 18362.1016
Thymio Suite 2.1.0-9cf577cd
Screencast of the bug: https://drive.google.com/file/d/1_E6DOc-CaaWrXSGheVA96JAhjdN0jDsE/view?usp=sharing
The issue seems to only happen for specific window dimensions

mbonani, 2020-08-28:
OK thank for the video I can reproduce the bug.
It append for me on the include QT browser when the window is full size of the desktop. It also append on Chrome when you play with windows size.

nbornand, 2020-09-03:
Great thanks. I've also noticed that the same behavior is observed for other blocks as well

mbonani, 2020-09-10:
fix in latest commit

closed 2020-09-10 by mbonani