Mobsya/aseba #850: VPL3 - Cannot inspect code sent to Thymio from Aseba Studio

nbornand, 2020-08-26:
**Setup**
Windows 10 Pro, version 1903 build 18362.1016
Thymio Suite 2.1.0-9cf577cd
**Repro steps**
Create a VPL3 program and load it on a Thymio
Connect to the same robot with "Watch from Aseba Studio"
Observed: The code is empty OR shows the last program sent with VPL 2.*
Expected: Should show the code corresponding to the VPL3 program

mbonani, 2020-08-26:
This behaviour was not implemented in VPL3, I will see with developers what we do. The generated code is not easy to read and follow as on VPL1.

nbornand, 2020-08-26:
Alright. Maybe worth checking with Francesco as he was expecting that to work.

mbonani, 2020-09-08:
this feature was add. Will be available in next release

closed 2020-09-08 by mbonani