Mobsya/aseba #868: robot not detected on vplv3 but work on legacy vpl

xens, 2020-10-10:
Running aseba 2.1.1 on Ubuntu 20.04, (with proper udev rules as stated in the doc).

The robot is detect on the main menu 

[image]

But not anymore inside vplv3

[image]

works fine with the legacy vpl

[image]

Is there a way to obtain more debugging info when launching the aseba suite to give you more input on this issue ?

thanks !

mbonani, 2020-10-10:
Hello,
Can you please try to update the firmware of the robot? Just click on the circular arrow.
Kind regards

xens, 2020-10-10:
I did the upgrade and restarted the robot (the circular arrow now disappeared) but vplv3 still marks the robot as "not detected", I tried with the USB cable instead and it's the same behavior, work in both cases (wireless+cable) with vpl but not in both cases with vplv3.

mbonani, 2020-10-10:
ok is it working with scratch? or blockly?

xens, 2020-10-10:
yep sorry should have tried before:

* works with Scratch
* works with VPL
* doesn't work with Blocky
* doesn't work with VPLv3

mbonani, 2020-10-12:
Ok, seems strange that it work with scratch but not with VPL3 or Blockly, all tree are web-based program and connection mechanism is the same. 
You can have more information about debug information launching Thymio Suite by command line:

`flatpak run org.mobsya.ThymioSuite`

 You can also launch before Thymio device manager by hand and have a look. 

`flatpak run --command=thymio-device-manager org.mobsya.ThymioSuite`

xens, 2020-10-12:
Ok I did a full charge of the robot over the week end (after the battery died) and it's now working with the 4 programs :sweat_smile: so I don't know what fixed it if it's the complete power reset of the robot, or the firmware update or both but now everything is working fine. Closing this issue, thanks for your support @mbonani

closed 2020-10-12 by mbonani