Mobsya/aseba #869: Build fail with error c2131 and error c2065

adrijasen1, 2020-10-10:
I did everything as mentioned in the build repo. but at the last step, my build fails with the following errors:

  C:\..\vcpkg\installed\x64-windows-static\include\boost/asio/uses_executor.hpp(50,26): error C2131: expressio
n did not evaluate to a constant [C:\..\vcpkg\aseba\thymio-device-manager\thymio-device-manager-lib.vcxproj]

  C:\..\vcpkg\installed\x64-windows-static\include\boost/asio/detached.hpp(52,22): error C2131: expression did
 not evaluate to a constant [C:\..\vcpkg\aseba\thymio-device-manager\thymio-device-manager-lib.vcxproj]

  C:\..\aseba\aseba\thymio-device-manager\wireless_configurator_service.cpp(27,106): error C2065: '_1': undecl
ared identifier [C:\..\vcpkg\aseba\thymio-device-manager\thymio-device-manager-lib.vcxproj]

  C:\..\vcpkg\installed\x64-windows-static\include\boost/asio/uses_executor.hpp(50,26): error C2131: expressio
n did not evaluate to a constant [C:\..\vcpkg\aseba\thymio-device-manager\thymio-device-manager-lib.vcxproj]

  C:\..\vcpkg\installed\x64-windows-static\include\boost/asio/detached.hpp(52,22): error C2131: expression did
 not evaluate to a constant [C:\..\vcpkg\aseba\thymio-device-manager\thymio-device-manager-lib.vcxproj]

I need help!! Thanks in advance.

mbonani, 2020-10-10:
Hello,
Yes it is a recent bug we fix using a older compiler in Visual studio.  You have to use an older compiler that you can install from the visual studio installer. Install 

MSVC v142 - VS 2019 C++ x64/x86 Build Tools (v14.25)

Then add ` -T version=14.25` in your cmake command. It should work.
I will try to update the build information

adrijasen1, 2020-10-10:
I still couldn't make it work. The errors are still the same.  I will try again and update if it works.  I am using Qt 5.12.5 and Boost 1_67_0. Do you think that might be the problem? 
And thank you for your quick reply.

mbonani, 2020-10-11:
use boost 1.72
Qt 5.12.5 should work

closed 2020-10-14 by mbonani