Mobsya/aseba #878: Cannot start with simulator

blackPantherOS, 2020-11-12:
I tried use with simulator, cannot start any programming environment.

https://youtu.be/6yCVJimMJdU

mbonani, 2020-11-12:
I there something that can block zeroconf/avahi on your system?

mbonani, 2020-11-12:
was it working with previous version?

blackPantherOS, 2020-11-13:
> I there something that can block zeroconf/avahi on your system?
I don't know about it...
[image]

> was it working with previous version?
i haven't tried the previous version sorry, but I will check later, if will help you..

mbonani, 2020-11-13:
Hello,
my avahi looks like that when I have a simulator with 4 robots:
[image]

So it confirm that the  problem is there, now why? it is difficult to say because I didn't master Linux and your distrib. 
No need to install older one. But you install it using flathub? Or you compile it yourself?

Kind regards

blackPantherOS, 2020-11-13:
>But you install it using flathub? Or you compile it yourself?

I didn't know this software before, I would like try from flathub repo only. 

The previous picture is not complete, this is good and same as your picture, but missing 'aseba.tcp'
[image]

mbonani, 2020-11-13:
The simulator describe itself on aseba.-tcp. "Thymio Device Manager (TDM)" is the one that manage real robots and make an entry point for the software to the local simulator. For sure if aseba.-tcp is missing TDM cannot found them and it is  why there are not shown in Thymio Suite.
I have no idea why the simulator is not describing itself on avahi. It seems that you have several IP address and I am not sure how zeroconf manage that. One difference from simulator and TDM it is that simulator describe itsefl only as local. On my computer I see them only on "lo IPv4"->local, and TDM I see them several time also like you on "eno1 IPv4-6"->local.
So did you have a firewall or something similar that block the simulator, is it a problem between ipv4 -ipv6? difficult to say.
What is you distrib based on?

blackPantherOS, 2020-11-15:
Solved (i needed some times for the investigate this issue)
So, my distro is same as my name... I build from source a new package (in RPM) and now everything is working fine. Seems the Flatpak package not full compatible with blackPanther avahi, therefore i will rebuild newer version. Thank for your time.
I will adding your suite to blackPanther official repository and our app-warehouse as well

[image]

closed 2020-11-15 by blackPantherOS

mbonani, 2020-11-17:
thank you for investigate