Mobsya/aseba #893: avahi-zeroconf do not work on debian with our flatpak?

mbonani, 2020-12-08:
could be related to this issue #878

lebardix, 2021-02-13:
Hello,
I have tested on Debian Buster but I not see "robots" in avahi.

An error is reported in flatpak 
i describe the error in forum 
http://aseba.wikidot.com/forum/t-13896509#post-4892411

`
/home/administrateur/.var/app/org.mobsya.ThymioSuite/data/mobsya/Playground/5be94b23/Thymio D3 sur primtux401/U0.DAT
/usr/include/c++/8.3.0/valarray:583: _Tp& std::valarray<_Tp>::operator[](std::size_t) [with _Tp = double; std::size_t = long unsigned int]: Assertion 'builtin_expect(i < this->size(), true)' failed.
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
Searching for "asebaplayground"
"asebaplayground" found: "/app/bin/asebaplayground"
qrc:/qml/ThymioSelectionView.qml:85: TypeError: Cannot read property 'icon' of undefined
`

Can you help me for this error ?

mbonani, 2021-12-13:
to be tested with latest master