Mobsya/aseba #898: Thymio does not start when there are too many events

guillaumev, 2021-01-01:
Hello,

We are trying to launch the attached program, which is supposed to write the name of my daughter THAIS. We have been trying the program as we were writing it and everything was working fine, until we added the last 3 steps.

If I remove 3 steps from our program, randomly, then Thymio starts again, however, as long as we keep the program complete, Thymio does not want to start... Could it be that our program is too big for Thymio ?

Thank you in advance for your help,

Guillaume.
[thais.zip](https://github.com/Mobsya/aseba/files/5759208/thais.zip)

mbonani, 2021-01-08:
Hello sorry for the delay, we will investigate. I will pass it to the VPL3 expert.

mbonani, 2021-01-08:
I confirm that the given program make a compilation error in aseba. Probably the size is too big

ypiguet-epfl, 2021-01-13:
Yes, the Thymio Device Manager reports "Script too big for target bytecode size". We'll have an error message soon, in a simpler wording and with translations. Your program helped us to do it, thanks a lot!
Maybe you could try to program directly in Aseba language (choose "Aseba Studio" in Thymio Suite) to have a more compact implementation.

guillaumev, 2021-01-15:
Thank you @ypiguet-epfl and @mbonani. We removed the middle branch of the A and it's working :-) Thanks for the suggestion to use the Aseba language, but I think it will be in a few years, my daughter is 6 years old for now, so probably too early for a textual language, but she is getting better and better in VPL3 !
And thank you for making Thymio ! She received it as a christmas present and has been playing with it pretty much everyday since then :-)

mbonani, 2021-04-23:
fix in latest release

closed 2021-04-23 by mbonani