Mobsya/aseba #919: Unable to open Thymio suite in scratch

chasset, 2021-04-21:
Hello,

After selecting on main screen "Scratch" and, on the following screen, "Thymio II" and push the button "Program with Scratch", the software opens a browser window at this URL: file:///private/var/folders/gg/vtdxvv_974l8p10gxg7hqrzr0000gn/T/org.mobsya.ThymioLauncher/tZNAXm.html

After that, the screen stays blank. May be it is missing a dependency?

Best regards,

Pierre-Olivier

mbonani, 2021-04-21:
Hello,
Could you perhaps try with another default system browser like Chrome?
Kind regards

mbonani, 2021-04-21:
Anyway the URL Inside the file is the correct one. I think this it is a protection mechanism of Apple/Safari

chasset, 2021-04-21:
It works! Many thanks

closed 2021-04-22 by mbonani