Mobsya/aseba #979: Issue to save/load in/from specific filename with VPL3 (Linux)

polochon777, 2021-12-28:
Hello,
With VPL3 when I try to save a program I don't have any popup to write a filemane and a location. I can save the program but only with "program.vpl3" in default `~/Download` directory. Then, I have to do a manual copy with my file manager if I want to keep it in another place.
This is almost the same to load a program, I can only load the program called `program.vpl3` in `~/Download` directory. Hopefully, I can drag & drop from anywhere as a workaround.

Thymio Suite 2.2 bf6e1c6

mbonani, 2022-01-10:
Hello,
cannot be related to #974 because it's a 2.3.0 bug. 
Can you try latest 2.3.0? 
Kind regards

polochon777, 2022-01-11:
Hello,
I confirm the issue is still here with 2.3.0 (after flatpak upgrade) :(

polochon777, 2022-11-14:
Hello, I update this issue because there is no issue with "pure" ubuntu distrop. This issue can be cloned with https://github.com/Mobsya/aseba/issues/1030 , because only primtux have the issue.