Mobsya/aseba #980: Communication wifi - Tymiosuite #débian

Eole35, 2022-01-02:
Bonjour à tous

Mon fils a eu un robot Thymio pour noël et je me suis empressé de lui installer.
alors le PC est un Dell E5430 avec la dernière version de Débian 11.2
Avec aseba en direct le Thymio s"associe nickel en wifi
Avec Tymio suite via Flatpak il ne communique pas en filaire ni en wifi,il ne le trouve pas

J'ai lu pas mal de topic, j'ai tout désinstaller et recomancer avec la procedure [https://github.com/Mobsya/aseba/blob/master/flatpak/README.md](url) et respecter a la lettre cette partie

_Set-up udev rules for thymio
Before being able to use a Thymio, you need to set up the appropriate udev rules to grant your user account permissions to use the device? This only needs to be done once per computer and doesn't have to be done again if you reinstall the Thymio Suite.
The instructs are given for a Ubuntu system. If you use another distribution, the steps to follow might be slightly different. Please refer to the documentation of your distribution concerning udev rules.
warning This step needs to be executed before you launch Thymio Suite.
warning You will need root access - if you don't own the computer, ask your network administrator.
Using nano or vi, edit /etc/udev/rules.d/99-mobsya.rules. For example

`sudo nano /etc/udev/rules.d/99-mobsya.rules`

Add the following line into the file:
```SUBSYSTEM=="usb", ATTRS{idVendor}=="0617", ATTRS{idProduct}=="000a", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0617", ATTRS{idProduct}=="000c", MODE="0666"
```
Save the file (Ctrl+x if you use nano), then execute

sudo udevadm control --reload-rules_

Maintenant ça marche nickel en usb
je me demande si il n'y pas un code a ajouter pour ajouter la clé et appairer le thymio?

En vous remerciant 
Loic

mbonani, 2022-01-10:
Bonjour 
il faut bien ajouter les deux lignes:

```
SUBSYSTEM=="usb", ATTRS{idVendor}=="0617", ATTRS{idProduct}=="000a", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0617", ATTRS{idProduct}=="000c", MODE="0666"
```
Meilleures salutations

mbonani, 2022-03-09:
Bonjour,

est ce que votre problème est résolu? si oui merci de fermer cette issue. 

Meilleures salutations

closed 2023-07-25 by Eole35