Mobsya/aseba #985: ThymioSuite 2.3.0 doesn't work on macOS 11.

ramuntcho, 2022-01-24:


mbonani, 2022-01-24:
can you be more precise? do you have a message ?

ramuntcho, 2022-01-24:
No message, just blank window

ramuntcho, 2022-01-24:
I use the application Apparency to test the certificate
[image]
[image]

ramuntcho, 2022-01-24:
Hi Michael,

I see that you are in Lausanne... and I am in Geneva !

mbonani, 2022-01-24:
seems you got not the correct version: 
[image]

get it there: https://github.com/Mobsya/aseba/releases/download/2.3.0/ThymioSuite-2.3.0-macOS.dmg

mbonani, 2022-01-24:
where did you dowload your version?

ramuntcho, 2022-01-24:
Hi,

It was downloaded on the official site :

https://www.thymio.org/download-thymio-suite-redirect/

It’s not correct ?

________________________________
Christian Oïhénart
Département de l'instruction publique, de la formation et de la jeunesse (DIP)
Service écoles-médias (SEM)
Secteur Logistique
Rue des Gazomètre 3
Case postale 241 – 1211 Genève 8
Tél: +41(0)22 388 63 76

mbonani, 2022-01-25:
Hello it is correct, but possible you get it at the wrong time, I reload a correct version last week. Can you confirm with a new download that all is working?
Kind regards

ramuntcho, 2022-01-25:
Hi,

I was using version 2.3.0 of December and I have just downloaded version 2.3.0 of January 17. Everything works and the application is correctly signed.
A remark because the preferences used by version 2.1.4 are ignored. Indeed, they were grouped in 3 containers (org.mobsya.AsebaPlayground, org.mobsya.AsebaStudio and org.mobsya.ThymioLauncher) whereas the preferences of the version 2.3.0 are simply in the Preferences folder.
Will this be the rule for the future versions of ThymioSuite? 

Best regards

mbonani, 2022-01-25:
Between 2.1.4 and 2.3.0 we upgrade qt versions and remake a ThymioSuite app the container. Perhaps difference came from there. But at same time Playground and AsebaStudio are full apps inside ThymioSuite that probably write there preferences when they are open. 
Normally 2.3.0  is the rule for next versions.

ramuntcho, 2022-01-25:
Ok, thanks for the quick answer