Mobsya/aseba #994: Where Thymio Suite get screen size on Raspberry Pi?

e-gaulue, 2022-02-27:
Following the documentation, we managed to get a working Thymio Suite on a Rpi 4.

But the application window size is bigger than the screen 1920x1080 (about 10%). Is there any way to set this in some env var somewhere or should we recompile?

Regards,

mbonani, 2022-03-09:
Hello,
Strange because minimum resolutions that is managed by  Thymio Suite is 1024x640.  Is there a "zoom"  in the settings of you your rpi4? like in windows?
we see that it is an issue on some other platform and I suggest to use 100% zoom

e-gaulue, 2022-03-09:
Well, I have the same feeling.

Out of the box, Rpi works but we (lots of Pi users) have big black borders on the screen. To avoid this, we set "overscan" and at reboot the display resolution is stil the same but the full screen is in use. I believe we are not far from a "zoom" idea. But I've no knowledge on how to get this information and how it's handled by RpiOS.

So without overscan : Environment 1920x1080 => use 1920x1080 of the screen but with borders.
with overscan : Environment 1920x1080 => use all pixels maybe 2100x1200 without borders.

As far as I understand Thymio Suite use Qt and get screen size from it. Looks it gets the number of pixel of the screen but not the resolution of the environment. It could explain what we observe. 

I also get a qt display GL error at startup. I have to find it back as I'm not behind it right now. It's maybe linked. 

Could we give ENV width/height parameters? It would be the easiest way because my assumption above may be wrong. If it's right, it would be necessary to get the environment size instead of the screen size.

Regards,

e-gaulue, 2022-03-11:
Great! That is the one we use.

Is there any chance to have one day WiFi communication with Thymio on Rpi 4? Everything I read, said "it doesn't work on Rpi3", but I don't see why it would then work on Rpi 4 as the board and the OS is about the same.

Regards,

mbonani, 2022-03-11:
On Rpi3 the problem is that the Wireless dongle of Thymio do not work because of an USB problem I never understand. It is not WIFI.