Mobsya/aseba #1030: Can't open file with blockly / ubuntu

polochon777, 2022-11-03:
Hello,
I can't open a previously saved .aesl program with blockly. The "open" button do nothing, and drag & drop from file explorer doesn't work.
This is annoying, the kids cannot save their program and reuse it the next programming session.

Note that the save button automatically save an "untitled.aesl" file in the ~/Downloads directory

OS is ubuntu 20.04LTS
Thymio suite 2.3.2-c408856c

Thanks a lot
Polochon

ypiguet-epfl, 2022-11-07:
Saving "untitled.aesl" into ~/Downloads must be due to the use of the default system browser. With the intern browser of Thymio Suite, the integration should be better with an improved user experience. In the initial view of Thymio Suite, can you click the top right button (three lines in a circle), unselect "Use your system default browser" and close the Tools panel by clicking the blue close button (left of "Tools")?

Depending on the system window manager (Gnome), it might happen that file dialogs are opened behind the main window. A notification popup should be displayed, but it can easily be overlooked. If in doubt, please reduce the size of the main window or move it around to check it doesn't hide the file dialog. The Blockly open button (folder icon, second from left) should display a file dialog also with the system browser. Which one are you using?

Thanks!

polochon777, 2022-11-08:
Hello,
I've retried with the main windows reduced or by moving around, nothing happend.
But your answer put me on the right path, this is probably an issue with the window system manager. 
In reality, the OS is not a "true" ubuntu but a linux distribution called [primtux](https://primtux.fr/ ) which is based on the current ubuntu version. I thought it was almost the same, but apparently not!
I've installed thymio suite on my personal computer (kubuntu 20.04.LTS), and everything  works perfectly.

I've also captured some logs on the console when I launch blockly:
```
[2022-11-08 13:07:50.091] [console] [info]   aseba_endpoint.cpp@L261:	Requesting list nodes( ec : Success )
file:///usr/lib/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
file:///usr/lib/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[2022-11-08 13:07:51.091] [console] [info]   aseba_endpoint.cpp@L261:	Requesting list nodes( ec : Success )
qrc:/qml/ThymioSelectionDeviceDelegate.qml:153:9: Unable to assign null to int
qrc:/qml/ThymioSelectionDeviceDelegate.qml:222:26: Unable to assign [undefined] to int
qrc:/qml/ThymioSelectionDeviceDelegate.qml:280:17: Unable to assign QQuickColumn to QQuickAnchorLine
Searching for  "asebaplayground"
"asebaplayground" found:  "/app/bin/asebaplayground"
ThymioDeviceManagerClientEndpoint::handleIncommingMessage Ping
[2022-11-08 13:07:52.091] [console] [info]   aseba_endpoint.cpp@L261:	Requesting list nodes( ec : Success )
```
Is it an issue with the QT version? I'm a little surprised, I thought that the goal of flatpak applications was to encapsulate everything in "one binary" to have a working application "out of the box".
But I confess I'm not competent on the subject. 

By the way, Primtux is quite popular for childrens and schools, it's a pitty than thymio suite is not fully working on (there is also other programming programs installed by default).
From your point of view, based on the logs, is it an issue in thymio suite or an issue on primtux side? I can contact primtux developers, but having solid technical inputs will probably helps to solve the issue.
Thanks a lot!